gev/reacthome-firmware #104: test for storing data in flash after turning off the power

DmitryTyazhlov, 2023-12-01:


DmitryTyazhlov, 2023-12-06:
[image]
очистка страницы размером килобайт занимает 32 милисекунды

DmitryTyazhlov, 2023-12-06:
запись килобайта во флэш на gd32f330k8 занимает 11 милисекунд
[image]

DmitryTyazhlov, 2023-12-06:
микроконтроллер без внешних плат и включения дополнительных внутренних переферий, при 3.1 вольте срабатывает прерывание, в нем очищается и записывается страница, емкость конденсатора 2100 мкф, время разряда с 3.1 до 2.6 вольт 48мс , до полного выключения микроконтроллера 158 мс, запись в память происходит успешно
[image]
[image]

DmitryTyazhlov, 2023-12-06:
конденсатор 2100мкф, порты микроконтроллера нагружены оптопарами от диммера 220в (все 12 каналов), в прерывании так же очищается страница и записывается килобайт
время от 3.1 до 2.6в 7.2мс, от 3.1 до выключения 24.6мс
при записи страница бьется (заполнена нулями)
[image]
[image]

gev, 2023-12-06:
и   GD32F450 нужно померить тоже, использовать, соответственно, его длину страницы

closed 2024-04-09 by gev